jsseng/arduino_robot #61: 'repeat' always requires 'times' keyword

jsseng, 2015-02-19:
I tried this, but it did not like it and required the 'times' keyword after the expression:

```
repeat (x<6) {
	x = x+1
}
```

ooee123, 2015-02-25:
Hmm, we can make a strong case for the word "while"

The current intent of repeat is for a loop to occur X times. Now, I can also imagine a case like this
```
repeat while (x < 6) {
   x = x + 1
}
```
That can be done. As a matter of syntax, for a while loop, would we prefer a repeat keyword or a while without a repeat keyword like this:
```
while x < 6 {
   x = x + 1
}
```

ooee123, 2015-02-25:
Oh, and
```
repeat (x < 6) times {
   x = x + 1
}
```
would let x < 6 evaluate in C style (resulting in the number 0 or 1), so that loop would happen exactly 0 or 1 times.
We can detect if the expression is a conditional expression, then one of two things can happen:
1) Warn the user that "Perhaps you mean to use a repeat while?"
2) From the detection of the conditional expression, convert that repeat into a while on our translated C.

I'm going to take the warn the user route for now, we can discuss.

ooee123, 2015-02-28:
OKAY, as per our discussion last Thursday 2/26:
```
repeat _expression_ {
}
```
Will evaluate as a while. Notice the missing keyword "times"
In addition, I included a warning if the conditional expression is not really a boolean. For example
```
repeat x {
   x = 4 < y
}
```
Will create a warning.

closed 2015-02-28 by ooee123