jsseng/arduino_robot #68: error with if condition

jsseng, 2015-02-23:
Given the following input:

```
var a = 2
if a = 1  {
	display "test"
}
```
I get the following error:

```
Caught Exception: expected '{', found '=' [Line 2]
```

ooee123, 2015-02-25:
Ahh, here is something.

Right now, equality can be done using any of these keywords
```
eq | equals | noteq | notequals
```
The "=" is reserved only for assignments.

True, we can detect if an assignment operator is used inside an if conditional, then warn the user. If we  wish, we can take it further and convert that assignment operator into an equality operator. I'm okay with leaving it with the eq, noteq, equals, and notequals keywords since we've cut != and ==, but I will welcome any input. Dr. Keen or Lana?

jsseng, 2015-02-25:
The only confusing part for me was supporting the <, >, <=, and >= operators, but then checking for equality requires the word 'equals' or 'notequals'.

aaronkeen, 2015-02-25:
We had agreed that '=' is reserved only for assignments *and* that assignments are statements.  As such, there should be an error if '=' is used inside the guard of an if.

closed 2015-02-27 by jsseng