jsseng/arduino_robot #74: _delay_ms() does not take variables as the argument

jsseng, 2015-02-28:
The built in _delay_ms() function only takes a constant as an argument.  When a variable is passed to sleep:

```
var x = 500
sleep x
```
then instead generate:

```
delay_ms(x);
```
This should call our own delay_ms() instead of the built in _delay_ms().  Do this only when a variable is the argument, otherwise call _delay_ms().

ooee123, 2015-02-28:
Where is delay_ms defined? How is delay_ms different than _delay_ms?

jsseng, 2015-02-28:
_delay_ms() is a built in macro that require a constant as an argument.

delay_ms() is something I haven't written yet, but it would just call _delay_ms() in a loop.

ooee123, 2015-02-28:
Oh, so that's how you could do that. I was wondering how to implement delay_ms.